MitjaNemec/Kicad_action_plugins #110: Replicate track linking 2 Layout

labodezao, 2020-11-26:
Hi Here, i have on my principal sheet and inside i have 2 hierarchical sheet that I duplicate (one called Trct.sch and Din.sch)

How I placed those 2 module on the PCB and I would like to replicate the tracks linking those 2 layouts. 

How can I do that ? Do I need to put that over anoter hieracical sheet ?

MitjaNemec, 2020-11-27:
Sorry, but I'll need a bit more info. Can you pate the schematics hierarchy so that I'll have a clearer picture of your project.

Something like this:
[image]

MitjaNemec, 2021-02-03:
closing due to lack of user input

closed 2021-02-03 by MitjaNemec