MitjaNemec/Kicad_action_plugins #111: Does the following indicate a MacOS problem?

jandyman, 2020-12-09:
When I try to run the script, the heirarchy shows up as empty:
[image]
But in the schematic editor I see this:
[image]
Does this indicate a problem with the MacOS version of the plugin?

MitjaNemec, 2020-12-10:
Did you save the schematics before running the plugin?

jandyman, 2020-12-10:
Yes, all changes were saved previously. Or, let's put it this way. If I open the design, and make no changes, I will see exactly what I have shown above. In other words, there are no unsaved changes at the time that I run the plugin.

MitjaNemec, 2020-12-10:
Just covering bases. Which KiCad version are you running? The plugin only supports 5.1.x

If you go to the project folder, delete and .log file and rerun the plugin and then hit cancel. There should be a `replicate_layout.log` file. Can you attach it here (you can edit it to hide personal data)?

jandyman, 2020-12-10:
It turns out there was already the appropriate log file in the project folder, and it should give some clues as to what going on
```
12-09 22:37:00 replicate_layout.action_replicate_layout 305:Plugin executed on: 'darwin'
12-09 22:37:00 replicate_layout.action_replicate_layout 306:Plugin executed with python version: '2.7.16 (default, May 14 2020, 09:34:58) \n[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)]'
12-09 22:37:00 replicate_layout.action_replicate_layout 307:KiCad build version: (5.1.6-0-10_14)
12-09 22:37:00 replicate_layout.action_replicate_layout 308:Replicate layout plugin version: 68 started
12-09 22:37:00 replicate_layout.action_replicate_layout 319:wx top level windows: wxWindowList: [<wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fa73c80a000> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fa73c13bde0> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fa73c92a800> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fa6da9d3fe0> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fa73b08fa00> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fa6ca472da0> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fa73cb4cc00> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fa6cec4ac50> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fa6cec7f920> >]
12-09 22:37:00 replicate_layout.action_replicate_layout 343:Preparing replicator with C3 as a reference
12-09 22:37:00 replicate_layout.replicatelayout 215:getting project hierarchy from schematics
12-09 22:37:00 replicate_layout.replicatelayout 217:Project hierarchy looks like:
{u'5FD3DB8D': [u'Pickup Preamp 2', u'/Users/andy/Documents/Bass and Music Tech/KiCAD schematics/Quad Preamp/file5FCEF305.sch'], u'5FCEF306': [u'Pickup Preamp 1', u'/Users/andy/Documents/Bass and Music Tech/KiCAD schematics/Quad Preamp/file5FCEF305.sch']}
12-09 22:37:00 replicate_layout.replicatelayout 225:getting a list of all footprints on board
12-09 22:37:00 replicate_layout.action_replicate_layout 370:source anchor footprint is u'C3'
Located on:[]
With filenames:[]
With sheet_id:[]
12-09 22:37:00 replicate_layout.action_replicate_layout 374:Corresponding footprints are 
[(u'C3', [])]
12-09 22:37:00 replicate_layout.action_replicate_layout 387:Showing dialog
12-09 22:37:00 replicate_layout.action_replicate_layout 391:Pcbnew position: wx.Point(101, 213)
12-09 22:37:00 replicate_layout.action_replicate_layout 397:Display 0: wx.Rect(0, 0, 3008, 1692)
12-09 22:37:00 replicate_layout.action_replicate_layout 401:Toolbar position: wx.Point(1986, 313)
12-09 22:37:00 replicate_layout.action_replicate_layout 408:Dialog position: wx.Point(1673, 313)
```
Notice the empty filename and sheet_id

MitjaNemec, 2020-12-11:
Whoa, this is certainly a new bug. Can you share the project? If you can, that would speed up the debugging, but if you can't then I'll need your help. I'll add instrumentation to the plugin, and you'll have to run the plugin again and send me back the .log file. Usually it takes 5 to 6 such cycles

jandyman, 2020-12-11:
Sure, I can share the project, it is attached. I can also help. I'm assuming it is not possible to debug the script in Pycharm or Spyder? That is how I usually develop Python code and it would certain make things easy.
[Quad Preamp.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/5680494/Quad.Preamp.zip)

MitjaNemec, 2020-12-11:
Well as they say, teach a man how to catch fish, ...

So if you are not in a hurry and you'd like to solve this on your own I would be glad to help how to setup up youre environment, how to start and how to fix.

As I don't really know what is the KiCad python environment on macOs, you should first find the path of the KiCad interpreter. Run pcbnew, open python console and run `import sys` and `print(sys.executable)`. 
Then you start pycharm open the folder where the plugins are installed and set pycharm up so that it will use KiCad's python interpreter.

I try to split GUI and backend (business logic) in my plugins so you should be able to reproduce this without KiCad and completely within pycharm.

Put your project in a subdirectory within replicate_layout folder and then you modify `main()` within `replicate_layout.py` so that it will start to replicate your board. You should be able to trace the code execution and stepping through the code. My first suspect is the `get_mod_by_ref()`. It probably does not find the "C3". Then you'll need to find why it does not find it,

If it is too much for you, just say so and I'll start working on it.

BTW, any comments that you add to the code which would improve readability are greatly welcome (there is certainly one missing in `get_mod_by_ref()` when it returns "None". Why and when does this happen.

jandyman, 2020-12-11:
Perfect, it all makes sense, I'll get to it later today.

jandyman, 2020-12-11:
OK, there are couple problems trying to follow the instructions. First, print(sys.executable) doesn't return the python interpreter:
```
print(sys.executable)
/private/var/folders/s9/85tgnyyd2ns4cdyyd4dlxg9m0000gn/T/AppTranslocation/4E9C081C-902C-43C1-A97F-477E66F98204/d/kicad.app/Contents/MacOS/kicad
```
Apparently this doesn't work "embedded interpreters":
https://stackoverflow.com/questions/5927633/how-to-get-the-current-python-interpreter-path-from-inside-a-python-script

So I tried looking off os.\_\_file\_\_, and found what looked like an interpreter:
[image]
But that doesn't work either, pycharm tell me this:
[image]
Why it thinks this is Python 3.9 is beyond me. So I tried just using a generic 2.7 interpreter, but then I realized that I didn't understand these instructions:
```
Put your project in a subdirectory within replicate_layout folder and then you modify main() 
within replicate_layout.py so that it will start to replicate your board. 
```

jandyman, 2020-12-12:
I'm wondering whether just for now, it would be easier to just debug with print statements, so to run this script from the scripting console on my project, what would I do. It looks like I modify main() but how and where do I need to put my project files.

MitjaNemec, 2020-12-12:
I've heard that the macOs does some randomization of executable paths, so the path that was reported by `print(sys.executable)` is random and probably changes every time you run KiCad.

From what I've heard the KiCad on macOs bundles python interpreter the same as Windows release. So can you look into the folder where you have KiCad binary installed and look for python interpreter. On Windows I can find it under `C:\Program Files\KiCad\bin\python.exe`. This is also the interpreter that I've added to Pycharm and `C:\Program Files\KiCad` is the Kicad installation folder.

Why you need to do this? If KiCad is using its own bundled python interpreter then `pcbnew` module is accessible (and maybe even compatible) only and with this interpreter. If KiCad uses system python (on Linux) then you can open default python console and import (and use) `pcbnew` module. And I don't know what is the situation on macOs.

So if you bear with me and we find this out I am pretty certain it will help me (and others) in the future

jandyman, 2020-12-13:
I'll keep working on trying to be able to actually debug with Pycharm, but if you could tell me how to run the script from the scripting console (it's probably simple), I could modify the code and debug with print statements. It probably wouldn't take too long, especially since you already have a hunch about where the problem might be.

jandyman, 2020-12-13:
Update: I tried a few things to get a debug environment set up in pycharm, with no luck. I cannot find a sensible place where a python environment is set up by kicad. It looks like kicad is doing weird things with that on MacOS, and although I'm a decent python programmer, I'm not really a sysadmin so I don't have the patience or knowledge to deal with it. 

I also posted on the forum about how to get set up for script debug on MacOS - no answers.

So I tried to set up a new environment using conda then installing this:
```
pip install kicad-python 
```
I was hopeful, but when I try to debug replicate-layout.py there are weird errors from within the code that imports kicad-python. It looks like once again they are doing weird stuff.

Also, I looked at main(), and it appears to be hardwired to test a project you've created, it even seems to reference specific parts in the design. 

In summary, if I'm going to go after this with debug by print statements (which I'm willing to do). I need two things

- A simple explanation of how to run the script from the scripting console.
- An explanation of how to modify main() to attempt to process my project

MitjaNemec, 2020-12-16:
Yeah, the main is hardwired to my test project. What you could do is to "rewire" it to your test project and then run the script. But it is the second part (run the script) that we have a problem with. You could just run pcbnew, open scripting console and run the plugin [https://thecodingbot.com/how-to-run-scripts-from-the-python-shell/](https://thecodingbot.com/how-to-run-scripts-from-the-python-shell/).

And I prefer to use logger instead of printf, but that is just a personal preference

jandyman, 2020-12-17:
I know how to run python scripts from a console. What I don't know is how to arrange arguments to the script so that my project is processed, and/or how to "rewire" the script to process my project. I didn't see any KiCAD scripting console example in GoogleLand. But I'll see if I can figure it out. Logging is fine. What I really like are Python 3 Fast Strings.

MitjaNemec, 2020-12-17:
Ok,

so my `main` calls `test_file` and to debug you issue first couple of steps of test file. So I'd just change `main` to something like

```python
def main():
    os.chdir(os.path.join(os.path.dirname(os.path.realpath(__file__)), "subfolder_with_your_project"))
    
    board = pcbnew.LoadBoard('name_of_your_pcb_file.kicad_pcb')
    # get board information
    replicator = Replicator(board)
    # get source module info
    src_anchor_module = replicator.get_mod_by_ref(C3)
    # have the user select replication level
    levels = src_anchor_module.filename
    # get the level index from user
    index = levels.index(levels[0])
    # get list of sheets
    sheet_list = replicator.get_sheets_to_replicate(src_anchor_module, src_anchor_module.sheet_id[index])
```

michalelektryk, 2020-12-29:
It seems to be broader linux problem. I'm been using the plugin for a long time on my windows 7 machine but I decided it's time to move forward and installed linux. And now my menu looks like this: 
[image]

michalelektryk, 2020-12-29:
It seems to be general problem with scaling or something:
[image]

MitjaNemec, 2020-12-30:
@michalelektryk I suspect that yours is a different issue. What is you linux distro and desktop environment? The more verbose you are the better.

jandyman, 2020-12-31:
OK, the holidays are over so I'm getting back to this. I tried running the following in the scripting console, but I've still got basic issues:
```
Startup script executed: /Users/andy/Library/Preferences/kicad/PyShell_pcbnew_startup.py
import os
os.getcwd()
'/private/var/folders/s9/85tgnyyd2ns4cdyyd4dlxg9m0000gn/T/AppTranslocation/5829D7B9-6D33-4005-9ED4-6EF037D227E9/d'
replicate_layout
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'replicate_layout' is not defined
os.chdir('/Users/andy/Library/Preferences/kicad/scripting/plugins/replicate_layout')
os.getcwd()
'/Users/andy/Library/Preferences/kicad/scripting/plugins/replicate_layout'
replicate_layout
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'replicate_layout' is not defined
os.listdir(os.getcwd())
['replicate_layout_GUI.py', 'compare_boards.pyc', 'no_wxpython.pyc', 'Replicate_layout_error.log', 'replicatelayout.py', 'action_replicate_layout.pyc', 'compare_boards.py', 'multiple_hierarchy', '__init__.py', 'duplicate-replicate_layout.svg', 'action_replicate_layout.py', 'no_wxpython.py', 'duplicate-replicate_layout.svg.png', '__init__.pyc', 'replicate_layout_GUI.pyc', 'replicate_layout_GUI.fbp', 'version.txt', 'remove_duplicates.pyc', 'remove_duplicates.py', 'replicatelayout.pyc']
execfile('replicatelayout.py')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "replicatelayout.py", line 23, in <module>
    import kicad.pcbnew as pcbnew
ImportError: No module named kicad.pcbnew
```

michalelektryk, 2020-12-31:
My distro is arch and desktop enviroemnt is SDDM. if it matters I scaled default font size, I may check with defaults

michalelektryk, 2021-01-01:
To be clear plugin is fully functional, only GUI renders wrong

jandyman, 2021-01-01:
Just a question on my comment a couple days ago. I would have expected that the environment used by the scripting console would include kicad.pcbnew. Is this an issue with kiCAD Mac, and if so, do you know where I report kiCAD issues?

MitjaNemec, 2021-01-01:
@michalelektryk can you open a new issue for this, as this is clearly different issue than jandyman's. If you can also paste full Kicad info it would help. Also can you test any plugins from other authors?

MitjaNemec, 2021-01-01:
@jandyman yeah, it seems that the python interpreter that is running the script can not find the pcbnew module. Can you run
```python
import sys
print(sys.version)
```
so that we can be sure which python are you runing and
```python
import sys
print(sys.executable)
```
to find out where the interpreter is located.

Only then we might consider reporting this upstream. But I suspect that MacOS executable relocation is the cause of this. While Kicad should be able to work with this. So I assume MacOS KiCad packagers should have the most knowledge on this.