MitjaNemec/Kicad_action_plugins #113: Save/Restore layout conflicting with multiple modules derived from same schematic

NickKnatterton, 2021-01-05:
Hey, 

first things first: thanks for this great project enhancing the capabilities of KiCAD!
Recently I´ve been using the save/restore function a lot to build projects from "tested" building blocks. (and building a small library of these blocks..)
But on restoring a layout in one specific project (integrity of the .pckl + .sch data was tested in a dummy project before) an 
"max() arg is an empty sequence" error occured. With some added debugging info (see attached log).
[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/5771197/save_restore_layout.log)
I tracked this down to inconsistent module identifiers (do You still call them tstamp? Not sure what else..) in the schematic. 
Now the fun part is: The identifiers are correct in my original schematic, but KiCAD changes them when updating the annotation because these specific identifiers already exist. Now why would it do that? Because this new module is an adapted version of an older module which I partly reused but also use in that project in its original state. So both of them have a few parts with the same id, which KiCAD realizes and then changes; corrupting the identifiers for the .pckl .
This probably is a very specific problem to my way of working and excessively using this plugin.  But maybe writing it down saves someone the effort to find this out the hard way ;-) 
As far as I understand there is no easy way to fix this (aside from not re-reusing your circuits) but for this specific case I just diffed both .sch files and copied the new identifiers into the .pckl.
Any info is this will still be a problem with KiCAD 6 or was the identifier changed anyway?  If not I would start writing a script to automate the identifier update from the new .sch file which could be incorporated into the plugin. 

Cheers,
Nik

MitjaNemec, 2021-01-05:
Hi Nik,

Thanks for the feedback. It it nice to know that Save/Restore plugin received some outside testing, I was always suspicious that almost no issues were reported given the complexity. But I contributed this to lack of use.

Looking at the log file, I see that you've modified logging, but I can not see what is the problem and I've only got partial picture what are the symptoms. As the plugin will have to change significantly for V6 compatibility (and other plugins also) I don't have any motivation of hunting this down and solving it. But better error message should be generated so that the user at least knows what is going on.

Are you in a position to share the project so that I can reproduce the issue locally?

If not, can you make a local copy of the project, so that you'll be able to run the plugin on the same project again and again? You'll send me back the .log file, I'll add some instrumentation and after a couple of such cycles, I should be able to implement proper error handling.

NickKnatterton, 2021-01-06:
Hi Mitja, 

then your plugin is definitely used a lot more then you think ;-) It took me some time to understand all the requirements for successfull restore operations, but by now I have about 15-20 different circuit snippets which were all created using a specific KiCAD template for identical naming and consistent design rules. Also I created a specific symbol/part which is always used as the reference part and contains size/orientation info. 
This way most projects end up being a collection of building blocks, rather than "new" layout. 

But to answer your questions: 
Symptoms:  Restore Layout stops with an error "max() arg is an empty sequence" 

Unfortunately I cannot share the project and my tries to recreate the problem were all stopped by KiCAD actually handling conflicting timestamps really well.. Given the awkwardness of recreating this problem, it will probably never occur again :) 
I could reproduce the behaviour by changing the "path" value for any of the modules in the .pckl file. This results in the script being unable to match schematic and .pckl. Here´s a screenshot of the changes I made to the save_restore_layout.py to understand my logging output. 
[image]
I think the source of my problem was reusing one schematic file as a starting point for multiple snippets and not reannotating all parts. This leads to parts with identical timestamp when using the schematics in the same project. KiCAD recognizes that and changes the conflicting timestamps, which again stops the script from matching the modules. It even gives you an error (identical time stamps found) when updating the PCB from the schematic, which I overlooked earlier..

MitjaNemec, 2021-01-07:
Thanks for the feedback. I'll try to recreate the problem, by modifying .pckl file and if successful, I'll add proper error handling.