MitjaNemec/Kicad_action_plugins #117: Place Footprints "Schematic hierarchy too shallow"

matthewlai, 2021-02-06:
I am trying to use Place Footprints to do placement by reference, but I'm getting a "Schematic hierarchy too shallow" error. The schematics has been saved (I exited and reopened KiCad to confirm).

I have one root sheet with one subsheet, and I looked at the code. It looks like find_all_sch_files() tries to recursively load all subsheets, and returns the error if there is 0 or 1 subsheets not including the root sheet (it doesn't look like the root sheet is ever added to the dict?)

In my case I have a root sheet with one $Sheet ... $EndSheet section, that links to my subsheet, which seems reasonable.

But the script is checking for "if len(self.dict_of_sheets) <= 1:". What is it expecting?

Thanks!

MitjaNemec, 2021-02-09:
Hi, if you want to place by sheet then you need to have at least two sheets on the same level.

The plugin expects the `len(self.dict_of_sheets)` to be at least 2 or bigger. (if you want to place by sheet then footprint from e.g. sheet #1 will serve as anchor and footprint from sheet #2 will be placed

What exactly are you trying to do?

matthewlai, 2021-02-09:
Thanks. It makes sense that there needs to be two sheets for place by sheet, but I was trying to do place by reference.

I had one sheet with 16 LEDs, and I wanted to place 15 of them relative to the first one.

MitjaNemec, 2021-02-09:
I am the first one to admit that plugins lack documentation and it is also possible that some names (e.g. place by sheet) might be chosen better. So all sugestions are welcome

matthewlai, 2021-02-11:
But I was trying to do place by reference not place by sheet. Why was it checking for 2 sheet in that case?

MitjaNemec, 2021-02-11:
If you can reproduce it, can you attach the place_footprints.log file which should be in the project folder.

swilson, 2021-03-23:
I get the same thing when trying to place by reference. Here is all my log shows:

03-23 10:59:24 place_footprints.action_place_footprints 414:Plugin executed on: 'darwin'
03-23 10:59:24 place_footprints.action_place_footprints 415:Plugin executed with python version: '2.7.16 (default, Dec 23 2020, 16:11:24) \n[GCC 4.2.1 Compatible Apple LLVM 10.0.1 (clang-1001.0.46.4)]'
03-23 10:59:24 place_footprints.action_place_footprints 416:KiCad build version: (5.1.9-0-10_14)
03-23 10:59:24 place_footprints.action_place_footprints 417:Place footprints plugin version: 54 started
03-23 10:59:25 place_footprints.place_footprints 213:getting project hierarchy from schematics
03-23 10:59:25 place_footprints.place_footprints 219:getting project hierarchy from schematics

swilson, 2021-03-23:
Removing the check fixes the issue: https://github.com/swilson/Kicad_action_plugins/commit/672094d84edf187224ea3a845c75a21dfb9f698c

commit 6f0f229b (2021-08-17): Move the check for hierarchy depth to place by sheed codepath

Fixes: #117
commit 4d624c11 (2021-08-17): Add the option to replicate edge cuts items only when required

Fixes: #117
closed 2021-08-17 by MitjaNemec