MitjaNemec/Kicad_action_plugins #126: Replicate Layout Error List index out of range

BillyPocketNC, 2021-03-25:
[image]
This message seems a little vague.

This error popped up when I tried to replicate a circuit where one of the chips in the master/source layout had more pins than one of the other layouts. (I modified a heatsink pad to have the required Vias as pictured below)
[image]
Updating all the footprints resolved the issue.

replicate_layout log:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6201240/replicate_layout.log)

MitjaNemec, 2021-03-25:
Yeah, the error message should be better, but besides figuring out that some footprints don't match between source and destination layout I don't think I don't think I can do.

Thanks for the report and the log. It is much appreciated

BillyPocketNC, 2021-03-25:
No problem. Thanks for the work you put into these plugins. They have saved me hours.