MitjaNemec/Kicad_action_plugins #127: How can I create new nets?

stormyweathers, 2021-03-26:
Hey there,

I'm building my own scripts, and the documentation is quite sparse.  I hope you could help me out a bit.

I'm trying to create layouts 100% from script.  I know how to place footprints and interate over their pads, but I haven't figured out how to create new nets to connect the pads.  I know how to connect pads to existing nets, but not how to create new nets.

Do you have any advice?

MitjaNemec, 2021-03-26:
I don't know of the top of my head. So I'd start by look at the code of WireIt action plugin (I am not the author) and I'd also search the [forum.kicad.info](https://forum.kicad.info/) forum. I seem to remember that Qu1çk replied to someone trying to do something similar

stormyweathers, 2021-03-27:
Thanks for the tip!
The WireIt plugin gave me exactly what I needed.

closed 2021-05-12 by MitjaNemec