MitjaNemec/Kicad_action_plugins #129: Replicate Layout replicating Edge.Cuts

okelk, 2021-05-11:
I have a design with one part ( LED ) protruding the board edge.

Peplicate Layout insists on replicatating part of the board edge lines on the Edge.Cuts layer.
-this leads to overlapping lines tedious manual cleanup later.

I can imagine some cases (e.g. internal cutouts) where someone might actually replicate objects on Edge.Cuts, but I guess it's something you want to avoid more often than not.

A checkbox or some other solution to avoid replicating objects on Edge.cuts would be great.
Sorry if there's already some way and I missed it - keep up the good work! this plugin is great.

MitjaNemec, 2021-05-12:
Thanks for reporting. The more users use the plugin, the more likely we'll find all of the edge cases (pun intended).
And I aprreicate your comment regarding the internal cutouts as my first thought would be to just ignore items on `Edge.Cuts` layer. So I'll probably add another checkbox.

As for locking and 5.99, the most proable way would be to rather use groups and grouping instead of locking (see #123).

MitjaNemec, 2021-08-17:
Should be fixed in 4d624c11f0d69d343a29a3f854485581bf856359

closed 2021-08-17 by MitjaNemec