MitjaNemec/Kicad_action_plugins #130: Replicate layout fails to find pcbnew window (list index out of range)

W-Floyd, 2021-05-19:
I believe it's searching for the pcbnew window, and cannot find it, when it is in fact open. Running Arch Linux, KiCAD 5.1.10, latest commit on master branch.

[image]

MitjaNemec, 2021-05-20:
yeah, that is the case. It is really difficult to make this part of the code cross platform, cross distribution and robust. I don't have and issue with 5.1.10 on Kubuntu 20.4.

Thanks for reporting it. I'll look into this, but currently I really don't have the time. 
BTW when you do open pcbnew, what is the window title?

waterlubber, 2021-05-28:
I'm having the same issue on Arch Linux/KDE Plasma as well, with KiCAD 5.1.10 (installed as a system program in `/usr/`). 
[image]
Here's the window information from KDE:
[image]
In this case, the window title was: `Pcbnew — [Current Working Directory]`

jano2358, 2021-06-04:
I'm having the same issue with the placement plugin:
[image]

Also running it in Arch with KDE.

jano2358, 2021-06-04:
yeah! most definitely related to lib versions, I downgraded wxPython to 4.0.7.2-5 and now it works (I wasn't able to sort anything yet, but it doesn't fail.

MitjaNemec, 2021-06-04:
Thanks for reporting back. I've modified that part of code a bit and I've added additional debugging code, which should add some info to the log file.

If you can, pleas run it on the setup that you have now and on the setup that was known to fail and send me back the `replicate_layout.log` file for both situations. I'll have to get this fixed sooner or later

jano2358, 2021-06-04:
where will be it created? (the log file)
also, every time I try to run it it says:
[image]

like people commented here: https://github.com/MitjaNemec/Kicad_action_plugins/issues/117
I have only one sheet, but I choose to sort by reference nr.
the lines 218~220 are different than the ones in that comment at the end now...

MitjaNemec, 2021-06-07:
@jano2358 
The .log file is placed in the currently opened project folder.
And your last issue is with Place footprint plugin and is tottaly different issue. Please open another issue, so that I'll not loose track of it.
>the lines 218~220 are different 
lines in which file. different than what? I don't follow, so you'll have to be a bit more verbose

jano2358, 2021-06-07:
sorry if I wasn't clear enough.
the .log file is not created and I guess is not created because of the other issue 'Schematic hierarchy too shallow'. 
There is an open issue about this, which is #117 , because of that I didn't open another one.
In #117 the last comment has a link to a commit which says that by removing lines 218~220 the problem is solved. But these lines are already different in the current version of the plugin. 
Therefore, I still cannot run the plugin and give you the log.

jano2358, 2021-06-07:
If I add two empty hierarchical sheets to my schematic I can run the plugin.
But a couple of days ago, I was not remembering this issue and I updated my system without preventing pacman to update python-wxpython. Then I'm running it now with the latest wxpython lib (python-wxpython-1:4.0.7.2-1) and it works. 
[place_footprints.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6610879/place_footprints.log)
I'm attaching the log file, in case it's useful, But I think you can close this issue.
The #117 is still valid though.

transistorgrab, 2022-01-22:
I had a look at this today since it failed for me too. Since 6.0 is officially released now I thought I make my new project with it, but now your great script does not work. (it did in 5.x).

It seems to me that Kicad has broken the filter of the title where it says "pcbnew".

It now says "schematic editor" in English. But this is now a localized string. In German it says "Leiterplatteneditor".
So I think it would be better to filter on the frame name, wich is "pcbframe".

I could not test it as a fast proof of concept, since I'm not familiar how to get a plugin compiled so that Kicad accepts it.
Line 331 is the affected in my version of the plugin.

This was my idea for fixing it:
`tw_names  = [x.GetName().lower() for x in tw]  ## Kicad 6 broke the window titles, they are now localized... we need to use the frame names`
`_pcbnew_frame = [x for x in tw if 'pcbframe' in x.GetName().lower()][0]`

MitjaNemec, 2022-01-22:
Thanks for reporting, I was waiting for somebody to report it. Even with pre-6.0 the solution was not proper as it depended on language. For the time being, I can only ask you to replace the code with 
`_pcbnew_frame = wx.GetTopLevelParent(wx.GetActiveWindow())`
It should work. But I will not fix this in the current repo, as I am currently porting plugins, and I've discontinued the maintenance of existing plugins. Thanks for understanding.

If the fix does not work, you'll have to wait a couple of weeks until I release V6 compatible plugins (Archive3D models and PlaceFootprints are already released)

transistorgrab, 2022-01-22:
Thanks for the fast reply. However, it now fails at line 334 with:

`"AttributeError: 'BOARD' object has no attribute 'GetModules'"`
after `pcbnew.GetBoard().GetModules()`

Seems like there was changed something else under the hood. :(

MitjaNemec, 2022-01-22:
Are you on the version from the [5.99_test ](https://github.com/MitjaNemec/Kicad_action_plugins/tree/5.99_test)branch?

transistorgrab, 2022-01-22:
No, it's the release version:

Application: KiCad (64-bit)

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 13:36:27
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

MitjaNemec, 2022-01-22:
Sorry for ambiguous quastion. So I'll be more verbose.

Which version of the plugin are you using. The one from the master branch, or the one from the 5.99_test branch in the plugin github repository? From the error I assume you are using the version from the master branch. This version supports KiCad 5.1.x. Plugin from 5.99_test branch should work with Kicad 6.0.x

transistorgrab, 2022-01-22:
I downloaded this version: https://codeload.github.com/MitjaNemec/Kicad_action_plugins/zip/refs/heads/master
from the link it looks like the master branch.

transistorgrab, 2022-01-22:
I just tried with the "test" version linked above but this just throws a non-descriptive "list out of range" error.

transistorgrab, 2022-01-23:
Hi, I would love to help out here.

However, I cannot figure out how to debug a script, how to run it on my board from the Python console (or external editor). Do you know of any resource that explains how to do this?

I wasted some time now to find a tutorial or some step by step guide but had no luck. :(

MitjaNemec, 2022-01-24:
Thanks for the effort. It seems that the script is broken. So you'll have to wait until I release a V6 version of it. I am currently working on it, and I don't see any sense of fixing a thing that will be replaced in a couple of weeks.

As for debugging. The first thing would be to take a full error message, not only a `list out of range`. It should point you to the code location where the bug. Next thing would be to grab a .log file, so you can see how the plugin ran, and when it stopped.

As for debugging, specifically for my plugins, I've split the code to GUI code and actual bussines logic. And the bussines logic can be debugged via IDE (PyCharm, VSCode). And self testing code is already included. But this might be too much for a common user

transistorgrab, 2022-01-24:
Hi I found the log file, did look for in in the plug-in directory instead of the project directory. (using the 5.99 test version)
It points to a definite code line:

```
01-24 16:11:51 replicate_layout.action_replicate_layout 345:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\action_replicate_layout.py", line 342, in Run
    replicator = replicatelayout.Replicator(board)
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\replicatelayout.py", line 146, in __init__
    sheet_id = self.get_sheet_id_alt(fp)
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\replicatelayout.py", line 122, in get_sheet_id_alt
    sheet_id = self.get_path(module).split("/")[-2]
IndexError: list index out of range
```

KiCad itself only displays the last line as error message. :(
[image]

transistorgrab, 2022-01-24:
To me it looks a bit like that on Windows there are mixed path characters in use "/" and "\"?

MitjaNemec, 2022-01-25:
No it has nothing to do with file paths
Add `logger.info("model path: " + self.get_path(module))` line before the line 122, and run the plugin again, and look at the .log file. Footprint (module) path should have at least two items. Footprint ID and Sheet ID of the sheet where the footprint is placed.

transistorgrab, 2022-01-25:
Error is now with this newly inserted line. I don't even see a list in this line. also the line numbers are the same, but the code lines are different due to the newly inserted line. What? 
```
01-25 20:00:37 replicate_layout.action_replicate_layout 345:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\action_replicate_layout.py", line 342, in Run
    replicator = replicatelayout.Replicator(board)
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\replicatelayout.py", line 146, in __init__
    for fp in bmod:
  File "…/KiCad/6.0/scripting\plugins\replicate_layout\replicatelayout.py", line 122, in get_sheet_id_alt
    logger.info("model path: " + self.get_path(module))
IndexError: list index out of range
```

transistorgrab, 2022-01-26:
Hi, today I had another dig into this.
I found that there are entries in your `bmod` list, that don't return a path.
```
bmod[-88].GetPath().AsString()
'/00000000-0000-0000-0000-000061b8c737/00000000-0000-0000-0000-000061f14363'
bmod[-87].GetPath().AsString()
''
```
What I found was that these are footprints placed on the board that are not in the schematics like NPTH for mouse-bites. I would think these are safe to be excluded from the list.
After adding this `if` statement in `__init__` I could get it to replicate one of my hierarchical sheets successfully.
```
for fp in bmod:
    if fp.GetPath().AsString() == "":
        continue
```
However, I have two hierarchical sheets that are both twice used in the design. For the components of the second one it is not recognized that there is a target sheet.
Any idea on this?

MitjaNemec, 2022-01-26:
> What I found was that these are footprints placed on the board that are not in the schematics like
Kudos. I should have remember this as it poped up a month or two back in this or some other plugin. 

Again, congratulations for finding this out and for fixing it.

MitjaNemec, 2022-01-26:
Yeah it was reported in #145