MitjaNemec/Kicad_action_plugins #132: Replicate layout pluign Problem

PFC123456, 2021-06-07:
Hi, I use Replicate layout plugin, and encounter a problem:
[image]

MitjaNemec, 2021-06-07:
Did you save the schematics before running the plugin?

PFC123456, 2021-06-07:
> Did you save the schematics before running the plugin?

Yes, I did save the schematic before running the plugin.

PFC123456, 2021-06-07:
Here's the log file.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6608070/replicate_layout.log)

MitjaNemec, 2021-06-07:
Hmmm,

1. can you deselect and select the item in source hierarchy level. I've had some changes here and I don't recall testing it with such a simple project.
2. Would you mind sharing the project?
3. Would you mind updating your KiCad installation? I am always testing the plugin against only the latest KiCad version

PFC123456, 2021-06-07:
> Hmmm,
>
> 1. can you deselect and select the item in source hierarchy level. I've had some changes here and I don't recall testing it with such a simple project.
> 2. Would you mind sharing the project? [...]

Thanks for your reply. Here's my project file.
[test_05.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/6608685/test_05.zip)

I have tried running place footprints plugin in Kicad (5.1.5)-3 and (5.1.9)-1, it work well.
But in both Kicad (5.1.5)-3 and (5.1.9)-1, Replicate layout pluign doesn't work.

MitjaNemec, 2021-06-07:
@PFC123456 , your issue with replicate layout plugin should be solved now

PFC123456, 2021-06-07:
> @PFC123456 , your issue with replicate layout plugin should be solved now

No, the issue still exists. Do you know how to solve it?

MitjaNemec, 2021-06-07:
You did update the plugin with the latest version? If you did, can you attach the latest `replicate_layout.log` file that should be in your project folder.
I could reproduce the issue with your project, and I've implemented a solution which worked on my side. It could be that the issue is also OS specific (I am on Kubuntu 20.04)

PFC123456, 2021-06-07:
> You did update the plugin with the latest version? If you did, can you attach the latest `replicate_layout.log` file that should be in your project folder.
> I could reproduce the issue with your [...]

I update the plugin with the last version, and the issue is solved now (on Kicad (5.1.5)-3 / Windows 10)

Here's the latest log file. 
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6609635/replicate_layout.log)

I really appreciate your patience, reply, and support!

MitjaNemec, 2021-06-07:
No worries, thanks for reporting it

closed 2021-06-07 by MitjaNemec