MitjaNemec/Kicad_action_plugins #134: replicate layout error

sqrtmo, 2021-06-27:
Hello, 

it seems like I have problem trying to use the plugin with kicad 5.99 (attaching version info)
I did used the plugin before with older kicad so, hope I am not missing something stupid.

Any suggestions? 

Thanks in advance!

`Application: KiCad PCB Editor

Version: (5.99.0-10369-gb59bf064b4), release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.41.0

Platform: macOS Version 10.16 (Build 20F71), 64 bit, Little endian, wxMac

Build Info:
	Date: Apr 20 2021 20:24:27
	wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
	Boost: 1.75.0
	OCC: 7.5.1
	Curl: 7.54.0
	ngspice: 31
	Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
`
[image]

MitjaNemec, 2021-06-28:
Can you look for a `replicate_layout.log` file and attach it here. There should be additional info to help me debug this.

This is a recurring issue and I've not yet found the reason behind it. It looks like it might be a WxWidgets/wxpython version thing.

sqrtmo, 2021-06-28:
Mmm where can I find it? 
It si not in the plugin folder, nor I can locate it via locate in terminal.

MitjaNemec, 2021-06-28:
Sorry for not letting you know. It should be in the project folder.

sqrtmo, 2021-06-29:
Thanks a lot!
 
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6731435/replicate_layout.log)

MitjaNemec, 2021-06-29:
It seems you are running version of the plugin intended for 5.1 (from the master branch).

5.99 is supported only in the [5.99_test](https://github.com/MitjaNemec/Kicad_action_plugins/tree/5.99_test) branch

sqrtmo, 2021-06-29:
I missed to look in branches. Seems to work now. 

Thanks again and sorry for the noise.

closed 2021-06-29 by sqrtmo