MitjaNemec/Kicad_action_plugins #137: replicate layout not working 5.99_test

sqrtmo, 2021-07-30:
Hello, I am getting strange message when try to execute replicate layout plugin.

[image]

Attaching replicate layout log file. 
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6907674/replicate_layout.log)
`Application: KiCad PCB Editor

Version: (5.99.0-10369-gb59bf064b4), release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.41.0

Platform: macOS Version 10.16 (Build 20G80), 64 bit, Little endian, wxMac

Build Info:
	Date: Apr 20 2021 20:24:27
	wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
	Boost: 1.75.0
	OCC: 7.5.1
	Curl: 7.54.0
	ngspice: 31
	Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
`

Hope I am not making some stupid mistake.

Thanks in advance!

MitjaNemec, 2021-07-30:
Thanks for the feedback

Well I obviously still have a lot of python to learn. The cryptic message is because I assumed an `LookupError` exception could happen only when I raise it within the plugin backend. Obviously I was wrong. Maybe I should write my own exceptions for the faults that I handle, so that I don't use python's default exceptions.

Anyhow, I've added instrumentation, and I'd like if you can update the plugin and recreate the issue. The .log file should have some additional info.

sqrtmo, 2021-08-02:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6914723/replicate_layout.log)

Thanks a lot!

MitjaNemec, 2021-08-02:
I think, I've found a bug. The fix was pushed, so please try again and report the results back.

Thanks for testing

sqrtmo, 2021-08-02:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6915466/replicate_layout.log)

Hmm, I am getting error again.

MitjaNemec, 2021-08-02:
Can you share the project? it would speed up the debugging. But if you can't, Then I'll have to ask you for patience, as we'll need to do this back and fort a couple of times

sqrtmo, 2021-08-02:
[blueBoard_v2.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/6916613/blueBoard_v2.zip)

Yep, now the problem appears if I try to replicate TMC sheet, but If I delete all the sheets and then replicate TMC block it is working.

Thanks a lot!

MitjaNemec, 2021-08-04:
OK, I've found a bug. It was the UUID handling (the 5.1 compatibility fix striped leading or trailing zeros)

Please update the plugin, and try it on your end (I check only on MSW platform)

sqrtmo, 2021-08-05:
OK, it seems that it is working now. Tested on macOS.

Thanks a lot!!!

MitjaNemec, 2021-08-05:
Thanks for reporting and testing

closed 2021-08-05 by MitjaNemec