MitjaNemec/Kicad_action_plugins #139: Replication Plugin can't handle hierarchical sheet on hierarchical sheet

Dephrilibrium, 2021-08-16:
Hi everyone,

i noticed, that the _replication plugin_ can't handle a hierarchical sheet on other hierarchical sheet.

**Sheet structure:**
* Level 0: Main-Sheet
  * (Level 1: Channel 0 - This combination leads also to an error) 
  * Level 1: Channel-Collector
    * Level 2: Channel1
    * Level 2: Channel2
    * and so on...

**Description:**
I want to replicate 16 channels on a PCB. Due to space problems on the main-sheet, I started to collect them on a separate hierarchical sheet (channel-collector). But then, the _replication plugin_ can't handle the replication-sheets (ChannelX-sheets) anymore. Only an error with a **GUID looking** value appears (e.g. `b80b8bdc-da6a-4023-a383-9b30f1e0a67c`). The GUID remains the same on a retry.

It's working, if i put all channel-sheets onto the main-sheet (Level 0). The problems start to begin, when placing one or more channel-sheets on the channel-collector-sheet.

**KiCAD Version Info:**
Application: KiCad Schematic Editor (64-bit)

Version: (5.99.0-11727-g1ec70d30af), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.74.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 15 2021 20:59:36
	wxWidgets: 3.1.5 (wchar_t,STL containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.74.0-DEV
	ngspice: 34
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

MitjaNemec, 2021-08-16:
Thanks for reporting.

This might be #137. Check if you have the latest version of the plugin. If you do and if the replication still fails, then you've uncovered another bug.Then please search for the `replicate_layout.log` file which should reside in the project folder and attach it here, so I'll have more data.

Dephrilibrium, 2021-08-16:
Yeah, this seems to be the same error, but my _[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6992185/replicate_layout.log)_ has a little bit mor info than in #137:

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/6992185/replicate_layout.log)-Content:
> 08-16 14:06:52 replicate_layout.action_replicate_layout 305:Plugin executed on: 'win32'
> 08-16 14:06:52 replicate_layout.action_replicate_layout 306:Plugin executed with python version: '3.9.6 (default, Aug 12 2021, 01:34:31) [MSC v.1928 64 bit (AMD64)]'
> 08-16 14:06:52 replicate_layout.action_replicate_layout 307:KiCad build version: (5.99.0-11727-g1ec70d30af)
> 08-16 14:06:52 replicate_layout.action_replicate_layout 308:Replicate layout plugin version: 75 started
> 08-16 14:06:52 replicate_layout.action_replicate_layout 339:Preparing replicator with U2 as a reference
> 08-16 14:06:52 replicate_layout.replicatelayout 141:getting a list of all footprints on board
> 08-16 14:06:52 replicate_layout.action_replicate_layout 345:Fatal error when making an instance of replicator
> Traceback (most recent call last):
>   File "C:\Users/XMG/Documents/KiCad/5.99/scripting\plugins\replicate_layout\action_replicate_layout.py", line 342, in Run
>     replicator = replicatelayout.Replicator(board)
>   File "C:\Users/XMG/Documents/KiCad/5.99/scripting\plugins\replicate_layout\replicatelayout.py", line 159, in __init__
>     sheet_id=self.get_sheet_path(module)[0],
>   File "C:\Users/XMG/Documents/KiCad/5.99/scripting\plugins\replicate_layout\replicatelayout.py", line 115, in get_sheet_path
>     sheet_names = [self.dict_of_sheets[x][0] for x in sheet_id if x]
>   File "C:\Users/XMG/Documents/KiCad/5.99/scripting\plugins\replicate_layout\replicatelayout.py", line 115, in <listcomp>
>     sheet_names = [self.dict_of_sheets[x][0] for x in sheet_id if x]
> KeyError: '1b3cc5dc-0d08-4891-b6ad-90cff11a40df'

To test it easily, I've made explicitely an minimal example with the structure as shown below:
* Level0: Mainsheet
  * Level1: Channel-Collector
    * Level2: Channel0
    * Level2: Channel1

Level0 Mainsheet contains a connector (+,-,GND) and the Collector-Sheet
[image]

Level1 Collectorsheet contains two Channelsheets
[image]

And finally Level2 Channelsheet contains just 2 inverting OpAmps with a Potentiometer as URef
[image]

I hope, this will help!

MitjaNemec, 2021-08-17:
The bug should be fixed. Can you please confirm this on your end.

I am somewhat surprised that it took quite some time for bug to surface (the plugin couldn't handle sheets without any symbols with footprints on them (just subsheets and/or power ports) in 5.99 release

MitjaNemec, 2021-08-17:
I much appreciate the test case supplied

Dephrilibrium, 2021-08-17:
Yeah, i can confirm. Bug is fixed 👍
Also a combination like
* Level0: Main-Sheet
  * Level1: Channel0
  * Level1: Channel1
  * Level1: ChannelCollection
    * Level2: Channel2
    * Level2: Channel3
is working correctly 😁 
Thank you very much for the fast fix! You helped me a lot!

**Attention** for everyone who is reading this issue-thread!
I had to **completely restart KiCAD** after unpacking the files into the plugin-folder to get the plugin working. A simple reload in the _Action Plugins_ options within the KiCAD-Preferences wasn't enough!

Greetz,
Deph

closed 2021-08-17 by Dephrilibrium

brhoads, 2022-04-29:
@MitjaNemec can you confirm that it is expected to _not_ be able to replicate "AmpCol.kicad_sch" (The "Level1: ChannelConnection" sheet in the above example)? In other words, replication of a hierarchical sheet that only contains other hierarchical sheets (and no footprints) is not possible?

This screenshot is from 6.0.4 and plugin v77. I was hoping to see "AmpCol.kicad_sch" as an option.

[image]

I poked around in the footprint objects and didn't find a way to link them back to the footprintless "grandparent" hierarchical sheet.

Alternatively, do you have any suggestions on a "dummy" part to include in the hierarchical sheet container? I suppose a generic test point would be enough to get a footprint into the list so the container sheet could be "found", but I'd prefer to not have unneeded test points floating around the board.

MitjaNemec, 2022-04-29:
Currently I don't have the time to go deep into this issue. In the meantime with V6.0.x you should test anny issues with the newes version of replicate layout plugin which is available  under KiCAD's PCM (plugin and content manager). Any issues shoud be reported in the new plugin repo https://github.com/MitjaNemec/ReplicateLayout

commit 38f8c91a (2021-08-17): avoid the dictionary lookup for the sheet with no footprints

The sheets without any footprints (only subsheets are present) are not present in self.dict_of_sheets. In 5.99 varsion the plugin gathers schematics info via fooprint.GetProperties methods (Sheetfile and Sheetname).

Fixes #139