MitjaNemec/Kicad_action_plugins #142: save_restore_layout can't work in kicad6.0RC1

HaydenHu, 2021-11-17:
KICAD6.0RC1 was released with the hope of fixing save_restore_layout plug-ins

It seems that the schematic API has been postponed to 7.0, I hope not to wait for the API, directly parse schematic bar, this plug-in is too useful.

MitjaNemec, 2021-11-17:
Well, I plan to port most of my plugins to V6. But I'll start with the simplest ones as I also have to figure out how to integrate plugins in the "plugin and content manager". So my current plan is to start porting the simplest plugins ([Length stats](https://github.com/MitjaNemec/Kicad_action_plugins#length-stats) and [Archive project](https://github.com/MitjaNemec/Kicad_action_plugins#archive-project)). Then I'll continue with more difficult  ones ([Place footprints](https://github.com/MitjaNemec/Kicad_action_plugins#place-footprints), [Replicate layout](https://github.com/MitjaNemec/Kicad_action_plugins#replicate-layout)) and the last come the plugins that deal with schematics ([Save/Restore Layout](https://github.com/MitjaNemec/Kicad_action_plugins#saverestore-layout), [Swap pins](https://github.com/MitjaNemec/Kicad_action_plugins#swap-pins) and [Swap units](https://github.com/MitjaNemec/Kicad_action_plugins#swap-units)). But before I start with the last batch I'll have to check with KiCad core dev team what is the state of the new python API. If it will be nearly finished, I'll wait for it. Otherwise, I'll start porting. Especially Save/Restore Layout as it only reads the schematics.

roykrikke, 2022-01-01:
May I ask what your progress is? Could you possibly use help with, for example, reviews?

perigoso, 2022-01-03:
`But before I start with the last batch I'll have to check with KiCad core dev team what is the state of the new python API. If it will be nearly finished, I'll wait for it. Otherwise, I'll start porting. ` @MitjaNemec

New API has been postponed to V7, V6 is released so all is stable to work on

MitjaNemec, 2022-01-03:
Yeah, I am started porting the plugins, but it is a slow process as I am short on spare time lately. So it is going to be a slow haul.

And the Save/Restore will mos likely be the last plugin, as I'll have to study how to parse the schematic files (as there is no API available)

roykrikke, 2022-01-03:
Can I be of any help?

MitjaNemec, 2022-01-03:
Thanks for offering.

What is you background? Are you familiar with python? The major effort for Save/Restore layout plugin is to refactor `SchData` class. The class parses the main all schematics files to get the schematics file hierarchy, so that we have a dictionary mapping schematics filenames to schematics UUID. For parsing files we should use one of the available s-expressions parsers. It might be smart to check which one makes more sense. Additionally the `get_sch_hash` has to be refactored to calculate the hash of schematics files, but the parts that change when using the file in multiple projects should not be taken into account then calculating hash (`L` `F0` `F 0` `AR` `Sheet` `LIBS` and `EELAYER` records)

roykrikke, 2022-01-06:
I know python a bit and have been committing some code to https://github.com/ivandokov/phockup I am certainly no expert. I'll have a look at the code and get back to you then. Can I reach you somewhere private? Discord for example?

MitjaNemec, 2022-01-06:
Thanks for the offer.

Before you start coding, you should start with preparing two (or even better three) KiCad projects where you use the same file (or a copy of a file) as a hierarchical page. Make sure you annotate the schematics in all projects. Then take a look at how the specific hierarchical sheet schematics file looks like. What changes if you reannotate the project? What changes if you include this file into another project.

For observing changes, make a copy of the file in specific points in time. So you'll end up with the same file but at different points in time:
1. file used only in not annotated original project
2. file used only in annotated original project
3. file used also in second project, second project not annotated
4. file used also in second project, second project annotated
5. file used also in second project, first project reannotated
6. file used also in third project, third project not annotated

and then you can look at the differences with meld, or winmerge.

These observations will let you know what parts of the schematics file change. These parts will have to be ignored when calculating hash. If you write down your observations, and also keep and archive of all the projects at different points in time for sharing, this would also help me, if you are not able to proceed forward.

Before you start coding I'd first look for s-expressions parser, and evaluate different options. From my memory you have [sexpdata](https://github.com/jd-boyd/sexpdata) and you could also use parsers by [realthunder](https://github.com/realthunder/sexp_parser) or [qu1ck](https://github.com/openscopeproject/InteractiveHtmlBom/blob/master/InteractiveHtmlBom/ecad/kicad_extra/sexpressions.py)

And only then you can start coding.

HaydenHu, 2022-03-30:
Hello, how is the fix? Can I refer to the parsing section of this plugin for kicad_sch?
https://github.com/ian-ross/kicad-plugins
@MitjaNemec

MitjaNemec, 2022-03-30:
I've got the plugin backend working. It took a bit longer as I've found a couple of bugs in KiCad and some of them have been are fixed since 6.0.3, while for remaining there are workarounds. I've managed to get around the parsing of schematics files so you have no code to help you. Sorry.

I expect that GUI part of the plugin should be in place in two weeks. So the plugin should be in PCM in three weeks or so.

MitjaNemec, 2022-03-30:
Current code is at: https://github.com/MitjaNemec/SaveRestoreLayout