MitjaNemec/Kicad_action_plugins #143: New issue with Replicate layout pluign: no destination sheets are displayed

pascal-schwab, 2021-11-26:
Hi, 
I stumbled over a new/other Issue with missing destination sheet options in the "replicate layout plugin",  like mentioned in Ticket #132. The Test-zip-file on form this ticket #132 is working properly on my environment (all recent revisions)
But my actual project is not working. Right now I don't know where this issue is coming from, project or plugin.

Project-zip, replicate-log and picture with the missing sheet options are attached.

Any Idea?

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/7607284/replicate_layout.log)
[image]
[RF-Switch.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/7607287/RF-Switch.zip)

MitjaNemec, 2021-11-26:
Hi!

Sorry to inform you, but the issue is not with the plugin. When using hierarchical pages, for layout re-use, this works only and only when multiple logical sheets use the same schematic file. In your case sheets `Sheet61A3A9AC` and `Sheet61A3B298` should use the same schematic file (either `file61A3A9AC.sch` or `file61A3B298.sch`). This is supported by KiCad and it is a preferred way to handle repeated circuits. This way when you edit one 'channel' the changes are automatically made in all channels. But the downside is that with V5.1.x you can not have different values between the channels. I think that V6 should solve this.

MitjaNemec, 2021-11-26:
BTW thanks for reporting, it might have been a bug, so it is better to overreport

closed 2021-11-26 by MitjaNemec

pascal-schwab, 2021-11-26:
Sorry for that, thank you anyway.