MitjaNemec/Kicad_action_plugins #148: Vias are only partially replicated

murp-C2D2, 2022-03-03:
I am using the replicate footprint plugin to replicate some MOSFET drivers in my KiCad project. It works well for footprints and filled zones. However, I have noticed some reproducible issues  when it comes to vias and tracks.

Kicad version 6.0.2
Plugin version 1.1.0 (assuming the latest version is installed using the plugin manger)

The layout that I want to replicate, is appended from other PCB and consists of several vias that I have included for heat dissipation to the copper fill regions. When I use the plugin, I choose a footprint, then  while choosing the options for replication, I notice that only few of the vias gets selected. 

[image]

Once the replication task is complete, only these vias gets replicated. Another minor issue that I notice is that the replicated part is rotated by 90 degrees. 
[image]

I have tried the same for other layout that I wanted to replicate. The issue is the same but the number of vias selected is different. It is sporadic how many vias gets selected from different (sub)layouts. 

Am I doing something wrong here and are there any limitations to the plugin that I am not ware of?

I really appreciate the work you have put into developing this plugin and I thank you for that. I find it very useful and time saving to be able to replicate the layouts for repeating parts using this plugin. Just wanted to do my part to provide the details. Hopefully this will help you in the next development cycle. 

Thank you!

murp-C2D2, 2022-03-03:
I played around a bit and figured out that it takes only those vias that lies in between the components. I am not sure if the boundary for replicating is defined by the extreme position of the footprints or the courtyard of the components. 

I placed a test pad in the extreme corner and now it considers the vias that falls within this region. 

A solution here would be to either explicitly specify a boundary using the courtyard or some other drawing layer and then replicate all that lies within this region.

Selecting the parts in plugin
[image]

After the replication 
[image]

MitjaNemec, 2022-03-03:
First things first, the correct repo for the V6 plugin is at: https://github.com/MitjaNemec/ReplicateLayout. I am keeping this repo open just until I manage to port Save/Restore Layout plugin.

As for your issue, I have an idea how to solve it:
1. I'll replicate items which are in the same group as the anchor footprint even if they are outside of bounding box defined by sheet footprints. Note that this might already work, but I haven't tested grouping functionality thoroughly enough to know how it behaves in such border cases
2. I maybe add another checkbox for zones to expand the bounding box