MitjaNemec/Kicad_action_plugins #149: Global labels

juliusPeters, 2022-03-19:
I tried "replicate" with Kicad 6.0.0 and everything seems to work fine. I copied a schematic two times and it creates the new schematics and annotates all new components. The tracks are replicated as intended.

My problem is, that although there are now 2 new schematics with individually annotated comnponents, I cannot edit inside these schematics individually. Everything I do is immediately done for the other schematics as well.

This is especially bad as I use global lables to connect the circuits inside the replicated schematics to the main circuit in the main schematic at the top of the hierarchie. As I cannot change the global labels in the replicated schematics individually, these connections are always connected to each other. 

As an example: each of the replicated schematic should have an output labled out1, out2 and out3 that should connect to out1, out2 and out3 at the main schematic. But the 3 outputs of the replicated schematics only can have the same label. Instead of out1, out2, out 3, it is only possible to name each of them out1 OR out2 OR out3. As a consequence, they connect together and connect to only the one connection wih the corresponding label at the main schematic. 

I cannot figure out any solution for this problem. Do I miss something or is this some kind of bug?

Thank you for yout help
Julius

MitjaNemec, 2022-03-20:
Hi,

> My problem is, that although there are now 2 new schematics with individually annotated components, I cannot edit inside these schematics individually. Everything I do is immediately done for the other schematics as well.

This is how the hierarchical sheets work.

To make most out of hierarchical sheets, you should not use global labels within them. You can use local labels and hierarchical labels to bring signals in and out of the sheet. With sheets that need to be galvanicaly on different potentials this also means that you should not use power symbols. You should bring the supply in using hierarchical labels.

BTW, V6 version plugin has a home at https://github.com/MitjaNemec/ReplicateLayout

juliusPeters, 2022-03-21:
Thank you for your quick answer.
I do not exclude that I seriously misunderstand something, but:

As far as I understand there is no other way to connect sub-schematics to a super-ordinate schematic other than global labels. Hierarchical labels do not work for this purpose. They only work if the schematics are on the same level.

The root of the problem I described initially is, that when you copy a schematic, the copy has the same underlying scheet file - unless you change this manually.
Therefore, all changes within the copied schematic will apply to each copy, including global labels. This actually does not seems to be  consequent because the components are actually annotated differently. But this is obviously how it works.

To prevent this, you manually can change the underlying sheet file. This actualy is easy and Kicad even copies the content of the original sheet file to the new one automatically. When done, the copies can be treated separately and global labels are treated separately as well. They behave as if you create different schematics manually and not by copying them.

Unfortunately, when I tried replicate, it does not work when there is no underlying single sheet file for all copies. And I still do not understand how to connect multiple sub-schematics to the super-ordinate schematic with individual connections.

Sorry when I do not understand it ...

Julius

MitjaNemec, 2022-03-21:
Yeah, the plugin requires that all "channels" share the same schematics file. This guarantees that each channel is the same footprint wise and connectivity wise. I have no intention to change this, as it would mean a significant development effort. Also the plugin complexity would increase significantly and I expect that it would take quite some time to fix al the bugs, such a change and solution would introduce. Basically the plugin would duplicate a significant amount of KiCad's internal code, and I see no sense in this.

Obviously this places some restrictions how one proceeds to designing schematic in order to replicate layout using my plugin.

Again to solve the connectivity issues, you should bring the signals out of the sheet using hierarchical lables, and connect them in the parent schematics.

juliusPeters, 2022-03-22:
Ok, I understand.
Thank you
Julius

closed 2022-03-22 by MitjaNemec