MitjaNemec/Kicad_action_plugins #154: Save/restore layout : destination hierarchy does not match source level issue

roboteux, 2022-10-14:
Hi,
I’m using kicad 6.0.8 on windows 10.
I’m trying to use the save/restore layout plugin.

but when I try to restore the layout I get this error :
Destination hierarchy : [‘File: alim5V.kicad_sch’]
does not match source level: 'alim5V.kicad_sch

To use the plugin, I started in my source file here I have a sub sheet (alim5V.kicad_sch). In pcb new I’m selected one component and called save layout. that par worked and I have now a .pckl file.
Then in my other project : I’ve copied the file alim5v.kicad_sch. I’ve create a subsheet and linked it to my alim5V.kicad_sch file.
Then I’ve updated my PCB.
Then in pcbnew I’ve selected the same component has when I’ve save the layout. and called restore layout. And I get the error.

And now I’m stuck.

MitjaNemec, 2022-10-14:
Thanks for reporting.

Can you please attach the `save_restore_layout.log` which should be in your project folder.

roboteux, 2022-10-14:
here it is
[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/9784508/save_restore_layout.log)

I'll try to prepare a standalone kicad file that I can upload to show the issue.

roboteux, 2022-10-14:
While trying to reproduce the bug in a smaller board, I've found another issue

in :
def get_sch_hash(sch_file, md5hash):
    # load sch file
    with open(sch_file, 'rb') as f:

The variable sch_file, contains : C:\cartes_pcb\alimtest\File: alim5V.kicad_sch
The file name is correct, the path is correct, but the File: in the middle is not.
![Captur
[image]
e](https://user-images.githubusercontent.com/101811270/195816632-b824fa6c-14db-4c3c-9732-45e34c961e28.JPG)

For now I'll try to patch the python file to generate my test file

roboteux, 2022-10-14:
Actually,
this issue is the bug I was having to restore the layout.

So I've added this line in the python code : 
def get_sch_hash(sch_file, md5hash):
    sch_file = sch_file.replace("File: ","")**
    # load sch file
    with open(sch_file, 'rb') as f:

And now I can save my layout and restore it.
I've also attached the kicad file, that I use to do the save layout, in case that could help to figure out why the "File :" is present in the middle of the filename.

[alimtest.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/9784754/alimtest.zip)

MitjaNemec, 2022-10-14:
Hmm, inspecting .pckl, I see that "File: " string is already stored when saving layout. If you can re-save the layout and attach the .log file, this should highlight the issue a bit

roboteux, 2022-10-17:
Hi,
I've remove my patch, and this time I was able to save the .pckl file.
But I still can't import it (same error as the original post).
[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/9799925/save_restore_layout.log)
[alimsct.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/9799929/alimsct.zip)

I've attached the .pckl and the log

MitjaNemec, 2022-10-17:
Did you get the same error with new saved design (.pckl). I assume you get a different error. But I'd like to be certain, and if possible I 'd also like to get the .log file when you try to restore and you get an error.

MitjaNemec, 2022-10-17:
Yeah, the issue is that footprints (namely AU2) have the `Sheetfile` property is set to "File: alim5V.kicad_sch". This is the first time I've seen this. How was the layout file created? Since I don't have your footprint libraries, I can not delete the layout and recreate it from schematics. Do you also have official Kicad libraries? If you do, can you create a new project and only use official symbols in schematic, and create a layout file. Then check the footprint properties. using the following console code

```
import pcbnew
b = pcbnew.GetBoard()
brd = b
fp  = brd.FindFootprintByReference('AU2')
fp.GetProperty('Sheetfile')
'File: alim5V.kicad_sch'
```

MitjaNemec, 2022-10-18:
Ok, I can confirm that KiCad as of 6.0.8 (and maybe earlier - I'll have to check this) generates layout from schematics, where footprint property "Sheetfile" has prepended string "File: ". I'll get on this by further inquiring when and why was this change pushed and I'll obviously add code to properly handle this to all my plugins

MitjaNemec, 2022-10-18:
Again, thanks for reporting this and persisting

roboteux, 2022-10-19:
Thanks for your feedback. I'll be eagerly waiting for any new version of your awesome plugins.

closed 2022-10-19 by roboteux

MitjaNemec, 2022-10-20:
Opened an issue: https://gitlab.com/kicad/code/kicad/-/issues/12682

MitjaNemec, 2022-10-24:
Ok, as it turns out this is due to the bug in KiCad specifically in version 6.0.8. It is not a bug in the plugin.

As proper workaround would be quite ridiculous to implement, there will be no patch to fix this. The only thing I can do is to recomment that you switch to the nightly 6.0.x release. For windows these are available in https://downloads.kicad.org/kicad/windows/explore/6.0-testing and I can confirm that https://downloads.kicad.org/kicad/windows/explore/6.0-testing/download/kicad-6.0-testing_r26953.0593663462-x86_64-lite.exe does not have a bug anymore.