MitjaNemec/Kicad_action_plugins #155: [bug] Save/Restore Layout - File Name Handling

T0jan, 2022-10-20:
Hello,

While I tried to create restorable snippets of some of my modules I stumbled upon a bug which seems to have been introduced by a change in how KiCad handles filenames. On projects I generated with older KiCad 6 versions I still can save the layout as usual but as soon as I update the PCB from schematic, saving the layout will crash. I have a minimum example with log and error message to show whats not working:

[test.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/9828976/test.zip)
[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/9828979/save_restore_layout.log)
[image]

KiCad Version info:

Application: KiCad PCB Editor

Version: 6.0.8-f2edbf62ab~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.15.0-50-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Sep 30 2022 16:52:13
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.85.0
	ngspice: 36
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

MitjaNemec, 2022-10-20:
Yeah, this was already reported as #154 (and to eagerly closed by original reported) by @roboteux

There was a change in KiCad 6.0.8 how the footprint properties are encoded and it is throwing of my plugin (https://gitlab.com/kicad/code/kicad/-/issues/12682)

I plan to add code to handle this. I'll notify you here.

Thanks for reporting.

MitjaNemec, 2022-10-24:
Furthemore, as this is a repository for V5 version of plugins, I am going to archive this repository so it will be in read-only mode. Any new issues with V6 plugins should be raised in their corresponding repositories

closed 2022-10-24 by MitjaNemec